RoboTeamTwente/roboteam_ai #260: enterformation'

mrlukasbos, 2019-04-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**System**
 - Ubuntu 18.04 Bionic Beaver
-  ROS Melodic Morenia

ThijsLuttikhuis, 2019-04-10:
there is a static variable in the header of enterformation, but that should not be affected by two seperate entities of RTT_AI

mrlukasbos, 2019-04-11:
its a bug in robothub, I made a PR there

closed 2019-04-11 by mrlukasbos