RoboTeamTwente/roboteam_ai #296: Refstate switching on free kicks and penalties needs to be improved

rolfvdhulst, 2019-04-18:
On free kicks and penalties we need to run a seperate strategy to start it but at some point need to transition to normal play. Can be done through custom refstate but also a strategy that just succeeds.

mrlukasbos, 2019-04-19:
Some ideas:

### Idea 1 - Tree termination
Let a tree terminate and set a new tree (lets call it an 'after success tree') when the tree either succeeds or fails. This requires that free kick always succeeds after a given condition. 

**Pro's**  
Flexible when to switch to normal play

**Con's**  
Can be hard to implement and hard to maintain --> even more difficulties while making behaviourtrees
How to determine when you should switch?

### Idea 2 - Timing
Switch the tree after a given amount of second / ticks. 

**Pro's**  
Relatively easy to implement
Easy to maintain
Does not rely too much on behavior tree implementation to function properly 

**con's**   
Less flexible
Can give serious problems when playing real life 
How to determine how much time is correct?

rolfvdhulst, 2019-04-26:
Another idea would be event based - as soon as the ball has moved according to the referee's standards, keep running the old tree for some time, for example.

mrlukasbos, 2019-04-29:
fixed in #340

closed 2019-04-29 by mrlukasbos