RoboTeamTwente/roboteam_ai #553: `getRobotCommand(...)` in poscontrollers should have `avoidBallDistance` as a parameter

mrlukasbos, 2019-06-05:
to avoid obscurity in poscontrollers we should just always indicate what kind of poscontrol features we want in the request: when we want a command that drives to our position we should specify whether to avoid the ball or not etc. instead of setting them as class variables. (because class variables are remembered and can thus influence other skills)

rolfvdhulst, 2019-06-05:
So are you also making defense area's and others explicit here?

mrlukasbos, 2019-06-05:
hmm. thinking about this. The defence area usually as no edge cases except what we can do from the referee, so it's probably only about the referee. the `setCanMoveInDefenseArea` could maybe be removed then.  The only feature that really changes a lot is the ball avoidance distance. However, I can imagine that we sometimes also want to set a min-distance to defense area. I don't wanna make this complex so I'm still thinking of a neat solution

mrlukasbos, 2019-06-05:
related: #528

HaicoDorenbos, 2019-10-31:
So this issue should be closed, because it is related to #528 and #528 is closed?

rolfvdhulst, 2019-10-31:
As far as I can tell, the PR that closed #528 did not do anything related to this issue. I'm not sure what this issue even is about anymore though, so I can't really tell you if it's still relevant.

mrlukasbos, 2019-12-16:
This issue is related to rulesets. 

getRobotCommand takes some parameters as to what path it needs to take. Sometimes we actively need to avoid the ball. Currently I believe this is a boolean which either avoids the ball or not. I think it would be nice to have a variable distance to avoid the ball. With 0 being no avoidance. This is something I'd like because there are multiple ways we could avoid a ball.

During play you sometimes want to avoid it because it is passed to another robot, but you don't need to avoid it for a full meter then. Avoiding it with 30cm would be more than enough, while for ball placement you need to have a strict minimal distance of 0.7m I believe.

closed 2020-05-06 by RobotJesse