RoboTeamTwente/roboteam_ai #725: Console error after new push to main

Yuhanun, 2019-12-12:
@HaicoDorenbos 

Error: Access undefined field value in the FieldMessage class.

Happens every time when running without a world.

HaicoDorenbos, 2019-12-13:
This was not an error I introduced. In the past it did return 0.0 or some other magic value instead if some value was not defined (it was not possible to check whether a value was defined or not), because I improved the design we now can see whether a value has been defined or not :+1:. To solve this problem we have to change the inner design (which most likely takes a lot of time), such that no FieldMessage is created if it is not yet connected. 

In my point of view this is a low priority bug, because the system still works with this design choice. However it is indeed something that is messy and therefore if we have time left we will fix it.

rolfvdhulst, 2019-12-13:
This is very hard to solve as we also may just receive incomplete field messages (although quite unlikely unless people set up ssl-vision poorly). I do think we maybe should have some constructor that generates a field with the default values for A-division if they are not explicitly set by the message. Although this should still give warnings, it should not result in bugs.

closed 2020-05-06 by RobotJesse