RoboTeamTwente/roboteam_ai #726: ZMQPP error, port already taken

Yuhanun, 2019-12-12:
Sometimes when running roboteam_ai it causes an exception with zmqpp

what() : ...
... 127.0.0.1:.... port already taken.

Error is not exactly known to me right now as I cannot consistently reproduce it, will update issue when this happens again.

Yuhanun, 2019-12-17:
/home/john/git/roboteam_suite/cmake-build-debug/roboteam_ai/roboteam_ai
robot_output_target = GRSIM
[Roboteam_proto] Starting subscriber for world
[Roboteam_proto] Starting subscriber for geometry
[Roboteam_proto] Starting subscriber for referee
[Roboteam_proto] Starting subscriber for feedback_primary
[Roboteam_proto] Starting publisher for channel commands_primary
[Roboteam_proto] Starting publisher for channel settings_primary
[Roboteam_proto] Starting subscriber for world
[Roboteam_proto] Starting subscriber for geometry
[Roboteam_proto] Starting subscriber for referee
[Roboteam_proto] Starting subscriber for feedback_primary
[Roboteam_proto] Starting publisher for channel commands_primary
terminate called after throwing an instance of 'zmqpp::zmq_internal_exception'
  what():  Address already in use

This is a different error, however also important to fix.

mrlukasbos, 2019-12-17:
This is very weird. you are literallly trying to start two publishers to channel commands_primary.

mrlukasbos, 2019-12-17:
I just had this issue. I accidentallyhad two robothubs running on the same port so the error was rightfully thrown in that case.

closed 2020-01-24 by mrlukasbos