RoboTeamTwente/roboteam_ai #809: roboteam_world possibly stops when the number of robots change

rttwente, 2020-02-22:
**Describe the bug**
While running grSim, roboteam_world, and roboteam_ai, when the number of robots change in grSim from e.g 8 to 4, roboteam_world seems to stop sending messages to roboteam_ai. Only when roboteam_world is restarted, does it send messages to roboteam_ai again.

**To Reproduce**
1. Start grSim, roboteam_world, and roboteam_ai.
2. Confirm that everything works fine by moving a robot in grSim and seeing it reflected in the interface of roboteam_ai.
3. Change the number of robots in grSim, under Geometry > Game > Robots Count
4. Observe that this change in robots is not reflected in the interface of roboteam_ai

rolfvdhulst, 2020-02-23:
Have tried to reproduce but everything seemed fine. World correctly updates and AI receives it just fine. Can you reproduce this in any way? It could be a TON of things including 
bad roboteam_proto  something, 
AI somehow not updating interface correctly/deadlocking on a read, 
proto version mismatch with grSim, 
Changing the settings vision port accidentally,
 ZMQPP failing to read/write.

emielsteerneman, 2020-02-23:
I could also not reproduce this error with the latest grSim, roboteam_ai:development, and roboteam_world:development/master .. I'll check the tactics pc tomorrow

emielsteerneman, 2020-03-02:
Never observed again

closed 2020-03-02 by emielsteerneman