RoboTeamTwente/roboteam_ai #815: Some profiling information

mrlukasbos, 2020-02-25:
I have been playing around a little bit with my profiler to check the performance of AI. Some interesting things came up, which I would like to share. Please realize that at this point performance is not our greatest concern. (reliability and functionality is much more important, the tactics PC is quite strong). Also realize that the results may be different on other OS'es or other devices.

## tl;dr
- Decouple the interface and/or make it more efficient
- remove the maps from constants.cpp and make the either macro's or static constrexpr loose variables etc. those lookups can be done at compile time and should not take away approx 5 to 10% of calculation time... 
- remove the old world for an approx. 10% performance increase
- merge #802 for a 20% performance increase

### how did i profile?
- OS: macOS 10.15 "catalina" 
- Device: macbook pro 13" retina
- CPU (2,4 GHz Dual-Core Intel Core i5)
- RAM: 8 GB 1600 MHz DDR3
- GPU: Intel Iris 1536 MB

- All RTT libraries compiled with -O3 except for AI, which is compiled to -g -O0.
- start profiler after initial boot of rtt AI. ran usually around 3 minutes. (1 minute halt, 1 minute formation, 1 minute normal play) 

### What is the biggest performance hit?
It is by far the QT interface, which takes more than 50% of the CPU time, even during normal play. At least, this is the case on MacOS. Not sure what this percentage is on other devices, macOS is known for heavily prioritizing graphics. Still, the interface is very heavy. The most heavy parts are the graph widget and the visualization widget. 

on development, 20% of the CPU time went to the behaviour trees (this includes updating world) and 10% to smaller things (like protobuffers etc.). So where did the other 20% go?  It was IOManager. and more interestingly, on my branch (#802 Cleanup IOmanager) this was NOT the case. I suspect this is the price you pay for updating the world anytime a message comes in: you have to lock mutexes, wait a lot for them, and probably some cache misses because of switching threads etc. etc. -> on my device PR #802 shows a huge performance improvement of almost 20%! (I thought game analyzer was the bad guy here, but that only took about 1 percent)

### Updating the world
The new world updates quite a bit faster. it does the same calculations in about 70% of the time. Half of that time is looking up constants from constants.cpp, which are the maps for working geneva and working dribbler. These things are quick wins of course by removing the map. (maybe on O3 this would be optimized out, but i tested on -g O0)
Note that at this point the performance gain of the new world is not there, since the old one is still present. 

### Other insights
- The calculations are very nicely spread over 4 cores. 
- percentages are sometimes vague. 
- roboteam utils, roboteam proto and roboteam_world perform really good performance-wise

rolfvdhulst, 2020-02-25:
I would argue that profiling at -O0 is pretty much meaningless for AI. Function inlining, vectorization and other things affect performance heavily. Roboteam_World literally becomes almost 100-1000 times faster on my machine going from O0 to O3

mrlukasbos, 2020-02-25:
I did it on -O0 specifically because at the robocup we ran at O0 as well, so that we could catch mistakes more quickly would they occur.  Besides, although it is not entirely accurate, it definitely does give a representation of what is fast and what is not.

rolfvdhulst, 2020-02-25:
We did not run O0, we ran O1 (see e.g. the MRL branch).

closed 2022-09-15 by JAndringa