RoboTeamTwente/roboteam_ai #889: Segfault in Hungarian.cpp

RobotJesse, 2020-03-13:
Run development with 11 robots in grsim. For the other configs I tried (10, 9, 8, 6) it works fine. Dealer gets stuck in the Hungarian and oscillates between steps 3 and 5 a long time, then segfaults.
[image]

rolfvdhulst, 2020-03-13:
Interesting. I will attempt to investigate on Saturday. This is not our code by source so I might just build the algorithm myself and also document it properly. Can you also make this an issue on roboteam_utils?

rolfvdhulst, 2020-03-14:
I cannot seem to replicate it unfortunately. This is probably an edge case being missed in the code of the Hungarian.

rolfvdhulst, 2020-03-14:
If you can replicate it again try to see if you can get the exact call that causes it to fail. Then we can write that as a test.

closed 2022-09-15 by JAndringa