RoboTeamTwente/roboteam_ai #978: Field optional access

Yuhanun, 2020-03-30:
Currently field doesn't actually check whether an optional contains data, instead it just .value()'s it, we should check whether it contains data before doing so.

Yuhanun, 2020-03-30:
Specifically, here:
```cpp
    Vector2 lpl_begin = leftPenaltyLine.value().begin;
    Vector2 lpl_end = leftPenaltyLine.value().end;
    leftPenaltyPoint = lpl_begin + ((lpl_end - lpl_begin) * 0.5);
```

leftPenaltyLine seems to have not been assigned before

HaicoDorenbos, 2020-04-08:
Haico is a very busy person and to solve this problem properly you have to make huge design changes. Therefore it won't be my main priority now

rolfvdhulst, 2020-04-08:
Actually I propose a way to solve our field woes once and for all and that is just by 'normalizing' the data from SSL (e.g. filling out missing fields) and extrapolating whatever information it contains into a datatype field before we initialize it/make it globally available. This interpreted data type can then be easily used because there's no more optionals and other annoying things going on. Only downside is you will need a really complicated data intrepretation that checks for all cases of the incoming data and interprets it correctly.

HaicoDorenbos, 2020-04-08:
That is indeed the best solution I think, but since that takes a lot of time I will skip it for now